syssi/esphome-daly-bms #1: DALY BMS (k-type) support for ESPHOME ?

kirillch2000, 2024-10-17:
Unfortunately I couldn't find where else to write, so I'm writing to a similar project.

Please make DALY BMS (k-type) support for ESPHOME. Home assistant has a working project (https://github.com/patman15/BMS_BLE-HA) - but it's only for HA, and I'd like to use it together with OpenHab.

syssi, 2024-10-19:
@kirillch2000 I've prepared a first draft. Do you like to give it a try?

kirillch2000, 2024-10-19:
Thank you very much for responding to my request so quickly!

I tried your component - the result is as follows. All 3 of my DALY BMS (2 new - k-type and 1 old - last year's model) - work perfectly and give all the parameters! So this draft should be considered working!

output example:
Name  State  Actions
daly-bms-ble average cell voltage  3.2399 V  
daly-bms-ble capacity remaining  35.1 Ah  
daly-bms-ble cell count  16  
daly-bms-ble cell voltage 1  3.242 V  
daly-bms-ble cell voltage 10  3.239 V  
daly-bms-ble cell voltage 11  3.234 V  
daly-bms-ble cell voltage 12  3.241 V  
daly-bms-ble cell voltage 13  3.235 V  
daly-bms-ble cell voltage 14  3.241 V  
daly-bms-ble cell voltage 15  3.238 V  
daly-bms-ble cell voltage 16  3.240 V  
daly-bms-ble cell voltage 2  3.244 V  
daly-bms-ble cell voltage 3  3.237 V  
daly-bms-ble cell voltage 4  3.245 V  
daly-bms-ble cell voltage 5  3.239 V  
daly-bms-ble cell voltage 6  3.242 V  
daly-bms-ble cell voltage 7  3.241 V  
daly-bms-ble cell voltage 8  3.247 V  
daly-bms-ble cell voltage 9  3.234 V  
daly-bms-ble charging  OFF  
daly-bms-ble charging  OFF  
daly-bms-ble charging cycles  52  
daly-bms-ble charging power  310 W  
daly-bms-ble current  -6.70 A  
daly-bms-ble delta cell voltage  0.0013 V  
daly-bms-ble discharging  OFF  
daly-bms-ble discharging  OFF  
daly-bms-ble discharging power  0 W  
daly-bms-ble enable bluetooth connection  ON  
daly-bms-ble error bitmask  NA  
daly-bms-ble errors    
daly-bms-ble factory reset    ☐
daly-bms-ble max cell voltage  3.247 V  
daly-bms-ble max voltage cell  8  
daly-bms-ble min cell voltage  3.234 V  
daly-bms-ble min voltage cell  9  
daly-bms-ble power  310 W  
daly-bms-ble reset charging cycles    ☐
daly-bms-ble retrieve settings    ☐
daly-bms-ble state of charge  34 %  
daly-bms-ble temperature 1  10.0 °C  
daly-bms-ble temperature 2  10.0 °C  
daly-bms-ble temperature 3  11.0 °C  
daly-bms-ble temperature 4  11.0 °C  
daly-bms-ble temperature 5  NA  
daly-bms-ble temperature 6  NA  
daly-bms-ble temperature 7  NA  
daly-bms-ble temperature 8  NA  
daly-bms-ble temperature sensors  4  
daly-bms-ble total voltage  51.80 V

syssi, 2024-10-19:
Could you provide some more details about your DALY BMS models/versions? I would like to mention the devices as supported at the README. Could you tell me the BLE name of each device?

kirillch2000, 2024-10-20:
Hi. Yes, of course.
These are 2 new Daly Smart BMS K-Series 100A
https://www.dalybms.com/48v-lifepo4-bms-daly-k-series-smart-bms-3s-to-24s-40a-60a-100a-product/
The older one is probably A-series (looking at the manufacturer's website, I have not seen this mentioned anywhere else, in ESPHOME DOC this model is simply called DALY BMS (for UART)).
BLE names are those that are visible as their names? then - DL-xxxxxxxxxxxxxxxx (mac address) and Dxxxxxxxxxxxxxxxx
Contact me, I can scan their profiles and anything else if needed - I have this equipment at home.

syssi, 2024-10-21:
Did DALY change the protocol over time or does your old DALY BMS support boths protocol (the one with the `0xA5` preamble and the `0xD2` preamble)?

kirillch2000, 2024-10-21:
HI ! 
 Unfortunately, I have not studied the details of the old version protocol. It was important for me to connect new (k-type) versions.
If there is such a need - can you make a trial version of the driver for logging and studying this issue and I can run it on a separate ESP32 with a connection to this BMS

syssi, 2024-10-21:
It's not so important. The mystery will be solved some day.

I've another request: I've implemented some switches and buttons:

```
button:
  - platform: daly_bms_ble
    retrieve_settings:
      name: "${name} retrieve settings"
    restart:
      name: "${name} restart"
    shutdown:
      name: "${name} shutdown"
    factory_reset:
      name: "${name} factory reset"
    reset_current:
      name: "${name} reset current"

# ...

switch:
  - platform: daly_bms_ble
    charging:
      name: "${name} charging"
    discharging:
      name: "${name} discharging"
```

Could you test these (excluding the `factory reset` and `reset current` button)?

kirillch2000, 2024-10-21:
Ok, I'll test these buttons tomorrow.

syssi, 2024-10-25:
Did you find some time to test the controls?

syssi, 2024-10-29:
Thanks for the hint! I wasn't aware of the PR. A lucky detail: The PR is focused on `uart` communication. This project is mostly about BLE communication.

syssi, 2024-10-29:
Interesting.. some registers are different: https://github.com/esphome/esphome/pull/7524/files#diff-7100260b820cb090bf4ad0f7d9d9b7fc105a49a0a5b4efd6287a3e70c474b33fR37-R48

kirillch2000, 2024-11-04:
> It's not so important. The mystery will be solved some day.
> 
> I've another request: I've implemented some switches and buttons:
> 
> ```
> button:
>   - platform: daly_bms_ble
>     retrieve_settings:
>       name: "${name} retrieve settings"
>     restart:
>       name: "${name} restart"
............................
> 
> Could you test these (excluding the `factory reset` and `reset current` button)?

Sorry for the delay - I was on vacation. I added these buttons - I certainly didn't do a factory reset :) - everything else looks like it's working - "BMS retrieve settings" I still don't understand how it works - there are no statistics updates, and the data is only received by the component interval (update_interval)

syssi, 2024-11-04:
The settings will he just retrieved, decoded and printed to the log.

felipecrs, 2024-11-11:
> Interesting.. some registers are different: [esphome/esphome#7524 (files)](https://github.com/esphome/esphome/pull/7524/files#diff-7100260b820cb090bf4ad0f7d9d9b7fc105a49a0a5b4efd6287a3e70c474b33fR37-R48)

@patagonaa, did you base yourself on some manual to build https://github.com/esphome/esphome/pull/7524? Perhaps you can share it here so we can double check these discrepancies for the greater good. :)

patagonaa, 2024-11-11:
> Interesting.. some registers are different: [...]

Yea, I based my code on what someone linked here:
https://github.com/esphome/issues/issues/5476#issuecomment-2258525930

I also got some information from the DALY PC Software (BMSTool for H/K/M/S), but the docs I just linked seem to match the PC software (although a lot of things are missing in those docs).

Honestly, I assume the docs you have (in this repo) are from an entirely different BMS series, because the BMSTool is definitely working and some fields (like 0x3D) are definitely correct for my BMS.

Bummer that we both kinda did the same component (though, yours more towards BLE and mine more towards UART), we probably could've saved a bunch of work if we had worked together 😅

patagonaa, 2024-11-11:
Also, just because i stumbled upon it:
this
https://github.com/syssi/esphome-daly-bms/blob/d6673c37b79c7f7496f322257b9ffd1ccc0ff993/components/daly_bms_ble/daly_bms_ble.cpp#L287-L290

will never show discharging power, as `daly_get_16bit` returns a uint16, so it is always positive.

Looking at the windows software, (at least in the Modbus protocol) the power does not have a sign, so it is probably positive no matter if charging or discharging, making it kinda useless.

syssi, 2024-12-21:
@patagonaa Good point! I've fixed the issue. The value is calculated now and the unsigned power measurement of the BMS firmware isn't used anymore.

closed 2024-12-21 by syssi