syssi/esphome-daly-bms #9: Known requests

syssi, 2024-10-19:
See https://diysolarforum.com/threads/a-new-uart-protocol-for-a-daly-smart-150a-bms.86306/

```
# d2 03 00 80 00 29 96 5f
# d2 03 00 00 00 3e d7 b9
# d2 03 00 a9 00 20 87 91
# d2 03 00 c9 00 03 c6 56
# d2 03 00 cc 00 02 17 97
# d2 03 00 3e 00 09 f7 a3
# d2 03 00 57 00 0c e7 bc
# d2 03 00 d7 00 01 27 91
```

```
# d2 03 00 80 00 29 96 5f
0x0080, 0x0029 -> Read 41 settings registers

# d2 03 00 00 00 3e d7 b9
0x0000, 0x003e -> Read 62 status registers (62 registers = 124 bytes + header + crc)

# d2 03 00 a9 00 20 87 91
0x00a9, 0x0020 -> Read software version (32 registers = 64 bytes)

# d2 03 00 c9 00 03 c6 56
0x00c9, 0x0003 -> Read password (3 registers = 6 bytes)

# d2 03 00 cc 00 02 17 97
0x00cc, 0x0002 -> Read production date (2 registers = 4 bytes), YYMMDD00 = 0x14080F00 = 2024/08/15

# d2 03 00 3e 00 09 f7 a3
0x003e, 0x0009 -> Read some reserved signal registers

# d2 03 00 57 00 0c e7 bc
0x0057, 0x000c -> Read 12 GPS/GSM registers

# d2 03 00 d7 00 01 27 91
0x00d7, 0x0001 -> Unknown
```

syssi, 2024-11-19:
```
# https://github.com/patman15/BMS_BLE-HA/blob/main/custom_components/bms_ble/plugins/daly_bms.py#L40
# https://github.com/patman15/BMS_BLE-HA/blob/main/custom_components/bms_ble/plugins/daly_bms.py#L129-L147
>>> d2 03 00 3e 00 09 f7 a3
<<< d203120000000075300000ffff0000000000000000476e

00 d2
01 03
02 12
03 00
04 00
05 00
06 00
07 75
08 30
09 00
10 00
11 ff
12 ff
13 00
14 00
15 00
16 00
17 00
18 00
19 00
20 00
21 47
22 6e
```

syssi, 2024-11-19:
```
0x0080, 0x0029 -> Read 41 settings registers

>>> d2 03 00 80 00 29 96 5f
<<< d2035201f40c8000010010000600060001000100010000ffff0dde0e4208fc0898000000001388138875307530753075300028007d002800000028007d0028000000000000000000000c8000140000000000000057351d
```

patman15, 2024-12-31:
Hey @syssi, I just figured out that [Bulltron also uses Daly BMS](https://github.com/patman15/BMS_BLE-HA/issues/85) and their app issues a write command right at the beginning:
```
D2 10 00 D4 00 03 18 0C 1F 08 29 00 1D E1
                  ^^ time code 24-12-31 08:29:00
```
Interestingly, looking at your [stored documentation on the protocol](https://github.com/syssi/esphome-daly-bms/blob/main/docs/Modbus%20protocol.xlsx) writing to 0xD4 does not make any sense to me. Any ideas?

Further, since we both seem to invest quite some time into figuring out what BMSs do, interested in collaborating more and save some efforts? Ways to PM me are in my github profile, in case you are interested to discuss, get to know each other. :smile: 
Guten Rutsch!

syssi, 2024-12-31:
Sure! I will contact you via discord.

syssi, 2024-12-31:
This Modbus register description is not very accurate. In my experience register address range `0xCE...0xE5` is used for model specific stuff. For example the `balancer` can be enabled using register 0xCF:

https://github.com/syssi/esphome-daly-bms/blob/main/components/daly_bms_ble/switch/__init__.py#L21