syssi/esphome-daly-bms #25: Add Balancing switch request

leodesigner, 2024-11-06:
Hi,

Thank you for your work @syssi !
Just got this project working with Daly Balancer. (not BMS)
The only missing feature is Balancer turn on/off switch.
I cannot find it in the modbus registers unfortunately :(

Any suggestions where to look? Snoop BLE HCI communication ?

syssi, 2024-11-06:
Could you prepare a btsnoop capture using the Android app?

leodesigner, 2024-11-07:
> https://www.dalybms.com/download-pc-software/

The link to the balancer used: 
https://www.dalybms.com/active-balancer-lifepo4-16s-daly-3-to-24s-1a-smart-active-balancer-product/

Looks like these are responsible:

Balancer BT logs

Value: d20600cf0000aa56
Value: d20600cf00016b96

00cf register

I will test today.

syssi, 2024-11-07:
Could you provide the complete btsnoop capture? It can be easily anonymized. I would like to add it to the project.

leodesigner, 2024-11-07:
Not sure how to anonymise it. There is a lot of other devices and I was specifically targeting balancer on/off switch. 
The other datapoints (cell voltages, temperature) are captured perfectly, except delta cell voltage needs correction.
It shows 0.0007 V instead of 0.007 V.

syssi, 2024-11-07:
I've added the new switch entity. Please give it a try by changing the `external_components_source` to the feature branch `@add-balancer-switch`:

```
substitutions:
  name: daly-bms-ble
  device_description: "Monitor a DALY Battery Management System via BLE"
  external_components_source: github://syssi/esphome-daly-bms@add-balancer-switch
```

leodesigner, 2024-11-07:
Would be great to read the switch status too. 
At this moment it shows as OFF (at least in my quick implementation)

syssi, 2024-11-07:
Just for completeness: In best case we retrieve the state of the switch/register priodically so the state of the switch get updated if the balancer is changed by 3rd parties (Android app f.e.). At the momnent it's open loop because 0xCF isn't retrieved periodically atm.

leodesigner, 2024-11-07:
Yeah, this is downside, we could read it at the startup after reset. Not the best option, but could at least provide some initial state.

syssi, 2024-11-07:
On the long run we should read all or some delected settings registers periodically. The initial state of the `charging` and `discharging` switch isn't populated properly at the moment too because of the same reason.

leodesigner, 2024-11-07:
Update on my end: it works
!
The green led on the balancer stops and starts blinking accordingly.

There are several additional non important sensors could be added in the future as well:

- cell balancing current
- cell id from which energy is taken/added

syssi, 2024-11-07:
Could you create one issue per feature and provide as much details (registers, example data etc.) as possible? I'm happy about every contributed feature too. Please keep in mind we aren't allowed to break the BMS compatibility. We should introduce a second YAML example soon for the balancer and introduce a `device_type` option if we retrieve different registers per device.

commit f4b735be (2024-11-07): Add balancer switch (Closes: #25)
commit d6673c37 (2024-11-11): Add balancer switch (Closes: #25) (#26)
closed 2024-11-11 by syssi