syssi/esphome-daly-bms #30: Add status text sensor

felipecrs, 2024-11-12:
As in the Excel modbus manual:

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/felip/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/felip/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#467886" vlink="#96607D">

R | charge/discharge   status | 0--65535 | 0=stationary，1=charging，2=discharging
-- | -- | -- | --

</body>

</html>

syssi, 2024-11-18:
We could use this register to add the missing sign of the current sensor. Do you think there is a need for a charging/discharging indicator nevertheless?

syssi, 2024-11-19:
I'm talking about this sensor:

https://github.com/syssi/esphome-daly-bms/blob/main/esp32-ble-example.yaml#L93-L94

And want to mitigate this issue:

https://github.com/syssi/esphome-daly-bms/issues/1#issuecomment-2469140633

Is my understand correct: The DALY app shows always a positive current while charging & discharging?

felipecrs, 2024-11-19:
Well, my sensor is showing negative values:

[image]

felipecrs, 2024-11-19:
The power one is always positive:

[image]

felipecrs, 2024-11-19:
I'll try it now, unplugging my power.

syssi, 2024-11-19:
Thanks! I will prepare a fix to sign the power sensors.

felipecrs, 2024-11-19:
Unplugged the power:

[image]

felipecrs, 2024-11-19:
Confirmed to work now, after #32:

[image]

syssi, 2024-11-19:
Awesome! Thanks for testing!

closed 2024-12-21 by syssi