syssi/esphome-daly-bms #34: Bluethooth not connecting

gabriktm, 2025-01-16:
Hello Sissy!

Im trying to implement the new daly bms with esphome. I bought the esp32 wroom and im trying to connected to bms with ble

YES
[16:34:22][W][daly_bms_ble:173]: [F7:C3:E8:21:C6:68] Not connected
[16:34:32][W][daly_bms_ble:173]: [F7:C3:E8:21:C6:68] Not connected
[16:34:42][W][daly_bms_ble:173]: [F7:C3:E8:21:C6:68] Not connected
[16:34:52][W][daly_bms_ble:173]: [F7:C3:E8:21:C6:68] Not connected
[16:35:02][W][daly_bms_ble:173]: [F7:C3:E8:21:C6:68] Not connected
[16:35:06][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[16:35:06][D][esp32.preferences:114]: Saving 1 preferences to flash...
[16:35:06][D][esp32.preferences:142]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[16:35:12][W][daly_bms_ble:173]: [F7:C3:E8:21:C6:68] Not connected
[16:35:22][W][daly_bms_ble:173]: [F7:C3:E8:21:C6:68] Not connected

[image]

Im using your YAML script.

Can you help me?

syssi, 2025-01-16:
Please compile and flash this YAML as first step: https://github.com/syssi/esphome-daly-bms/blob/main/esp32-ble-scanner.yaml

It will print all detected DALY BMS devices to the ESPHome log. At the moment it looks like the BMS isn't advertised via BLE. Is another client connected probably (Android app?).

felipecrs, 2025-01-16:
You can probably infer the mac address through the bluetooth module label too.

> Daly Smart BMS K Series 100A advertised via BLE as DL-xxxxxxxxxxxx (e.g. DL-40D63C3223A2) or DL-Fxxxxxxxxxxxx (e.g. DL-F28BBED000824) where the MAC address is the xxxxxxxxxxxx

syssi, 2025-01-16:
Good point! In this case the MAC address is probably `7C:3E:82:1C:66:8C`.

closed 2025-03-12 by syssi