syssi/esphome-daly-bms #35: Error Message: no response received

davidbatoha, 2025-01-19:
Hello, I am using the UART .yaml script and I constantly get te error message: **[21:10:36][D][modbus_controller:040]: Modbus command to device=210 register=0x00 no response received - removed from send queue**

Can anyone help?

Things I tried to get it running:

- change the RX and TX (I looked it up and now it should be ok)
- logger level set to VERY_VERBOSE instead of DEBUG
- changed all the register types to READ instead of HOLDING

Here you have my .yaml file

substitutions:
  name: hausakku
  device_description: "Monitor a DALY Battery Management System via UART"
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-daly-bms"
    version: 1.0.0

esp32:
  board: esp32-s2-saola-1
  framework:
    type: arduino

wifi:
  ssid: *****
  password: *****

ota:
  platform: esphome

logger:
  level: VERY_VERBOSE #DEBUG

web_server:
  port: 80

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:
  reboot_timeout: 0s

# mqtt:
#   broker: !secret mqtt_host
#   username: !secret mqtt_username
#   password: !secret mqtt_password
#   id: mqtt_client

uart:
  - id: uart_0
    baud_rate: 9600
    tx_pin: ${tx_pin}
    rx_pin: ${rx_pin}

modbus:
  - id: modbus0
    uart_id: uart_0
    # flow_control_pin: GPIO12

modbus_controller:
  - id: bms0
    address: 0xD2   #0xd2
    modbus_id: modbus0
    setup_priority: -10
    update_interval: 10s
    command_throttle: 50ms

sensor:
  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} cell voltage 1"
    address: 0x0000
    register_type: read #holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    state_class: measurement
    accuracy_decimals: 3
    filters:
      - multiply: 0.001

  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} temperature 1"
    address: 0x0020
    register_type: read #holding
    value_type: U_WORD
    unit_of_measurement: "°C"
    state_class: measurement
    accuracy_decimals: 0
    filters:
      - offset: -40.0

  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} total voltage"
    address: 0x0028
    register_type: read #holding
    value_type: U_WORD
    unit_of_measurement: "V"
    device_class: voltage
    state_class: measurement
    accuracy_decimals: 1
    filters:
      - multiply: 0.1

  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} current"
    address: 0x0029
    register_type: read #holding
    value_type: U_WORD
    unit_of_measurement: "A"
    device_class: current
    state_class: measurement
    accuracy_decimals: 1
    filters:
      - offset: -30000.0
      - multiply: 0.1

  - platform: modbus_controller
    modbus_controller_id: bms0
    name: "${name} error bitmask 1"
    address: 0x003a
    register_type: read #holding
    value_type: U_WORD
    unit_of_measurement: ""

syssi, 2025-01-19:
Could you try this component instead: https://github.com/esphome/esphome/pull/7524

The Modbus example from this project is untested and probably incompatible.

davidbatoha, 2025-01-20:
Hello @syssi I can not get the .yaml compiled.
How should my .yaml look like using the example that you recommand me?

syssi, 2025-01-24:
Your configuration should look like this:

```
substitutions:
  name: hausakku
  device_description: "Monitor a DALY Battery Management System via UART"
  tx_pin: GPIO16
  rx_pin: GPIO17

esphome:
  name: ${name}
  comment: ${device_description}
  min_version: 2024.6.0
  project:
    name: "syssi.esphome-daly-bms"
    version: 1.0.0

esp32:
  board: esp32-s2-saola-1
  framework:
    type: arduino

    
    
external_components:
  - source: github://pr#7524
    components: [ daly_hkms_bms ]
    refresh: 0s
    
wifi:
  ssid: *****
  password: *****

ota:
  platform: esphome

logger:
  level: DEBUG

web_server:
  port: 80

# If you don't use Home Assistant please remove this `api` section and uncomment the `mqtt` component!
api:
  reboot_timeout: 0s
  
uart:
  id: bms_uart
  rx_pin: GPIO04
  tx_pin: GPIO05
  baud_rate: 9600
  data_bits: 8
  parity: NONE
  stop_bits: 1
  debug:

modbus:
  uart_id: bms_uart

daly_hkms_bms:
  - id: bms_id
    address: 1

sensor:
  - platform: daly_hkms_bms
    daly_hkms_bms_id: bms_id
    voltage:
      name: "Battery Voltage"
    current:
      name: "Battery Current"
    battery_level:
      name: "Battery Level"
    remaining_capacity:
      name: "Remaining Capacity"
    cycles:
      name: "Battery Cycles"
    temperature_mos:
      name: "Temperature MOS"
    max_temperature:
      name: "Temperature Max"
    min_temperature:
      name: "Temperature Min"
    max_cell_voltage:
      name: "Cell Voltage Max"
    min_cell_voltage:
      name: "Cell Voltage Min"

text_sensor:
  - platform: daly_hkms_bms
    daly_hkms_bms_id: bms_id
    status:
      name: "Status"

binary_sensor:
  - platform: daly_hkms_bms
    daly_hkms_bms_id: bms_id
    charging_mos_enabled:
      name: "Charging MOS enabled"
    discharging_mos_enabled:
      name: "Discharging MOS enabled"
    precharging_mos_enabled:
      name: "Precharging MOS enabled"
    balancing_active:
      name: "Balancing active"

```

felipecrs, 2025-04-16:
Actually, here's it now: https://github.com/patagonaa/esphome-daly-hkms-bms

closed 2025-04-16 by syssi