syssi/esphome-daly-bms #39: Problem with new BT +WIFI Modul

fat-fred, 2025-04-09:
First: thank you. The code works Perfect and ootb, but...

i have a new Daly BMS with the new Modul.
i can only connect with the ESP in the moment i Power on the BMS.
If it is than connected - all works until i reboot the ESP.

The Modul provided the MAC only a few Second. After this, i can not more connect (hidden?).
Also an android BT scanner can not more see it. With the smart BMS App i can always connect.

Teset also the Batmon-ha integration. Here the same. Can see the BMS if i restart it, but connection not posible (too slow).

I can make more tests, but dont know what to do now !?

syssi, 2025-04-09:
This sounds like a software issue of the BLE module located at the BMS. A BLE server stops advertising as soon a connection is established. If the BLE connections gets lost it should start advertising again after a connection timeout. I guess the Android app stops the BLE connection gracefully. If you move your Android device out of range you should see the same issue. Could you verify this assumption? This situation cannot be resolved on the client side.

felipecrs, 2025-04-09:
Check if you connected the BMS to Daly Cloud through Wi-Fi. If you ever connected it to Wi-Fi, try resetting the BT/Wi-Fi module.

fat-fred, 2025-04-10:
i'm connected to the cloud with wifi. so, how do i reseting the modul?

But as i said: The smart BMS app can always connect (main device is a IOS, Android is a test device)

fat-fred, 2025-04-15:
thank you. reset has worked, but the same with BT.

fat-fred, 2025-04-15:
Sorry. My Fault. 
Yesterday i thought the wifi was resetet. After this, i could connect with Bt and connection to cloud was not possibel.
But after a reset of the esp it was the same - no connection to bluetooth.

Today i would look again. BUT Bt are connected??? After reset no connection, but - cloud has workt again?!

So i made some research:
@felipecrs you are right. With wifi connected there is no BT connection possibel.

But you only must delete it in the smart BMS App from the cloud and everything is working like it should :-)

so, thank you all for help.

closed 2025-04-15 by fat-fred

felipecrs, 2025-04-15:
That's great! This probably deserves a warning somewhere here:

[image]

felipecrs, 2025-04-16:
This Daly brand's support/website is so ridiculous. I could only find this:

https://www.dalybms.com/news/wifi-module-usage-instructions/

And there's no information whatsoever on how to reset.

I have the original "manual" here in my box. I'll scan it. But it's in Chinese.

felipecrs, 2025-04-16:
[image]
[image]

The photos are of really poor quality. If I ever find a scanner, I'll try scanning it.

I could not find anything on how to reset the module, but there's indeed a line saying that BT connection cannot be used once the BMS is connected to Wi-Fi cloud.

fat-fred, 2025-04-20:
They Write:

10. Local monitoring: In Bluetooth mode, when the WIFI module status is "offline" or deleted, a Bluetooth connection can be made through "local monitoring". The usage method is the same as the Bluetooth module.

To delete it you Must Go to the App in cloud Mode. Make a Long press on your Device. Then you can rename or delete it. After you delete it here you can Connect Everytime with BT